aanon4/BlueBasic #12: Serial/UART handshaking

mikewen, 2015-01-20:
Right now, BlueBasic only support hardware handshaking
SERIAL 9600,N,8,1,H

This will give error:
SERIAL 9600,N,8,1,N

Here are the codes in OS_serial_open function (os.c ) :
// Only support port 0, no-parity, 8-bits, 1 stop bit
if (port != 0 || parity != 'N' || bits != 8 || stop != 1 || flow != 'H')
{
return 3;
}

Also, I am getting 'Not in direct' error when I tried to write to serial port:
In Console:
BlueBasic CC2540/20150104225856/OAD v0.6
http://blog.xojs.org/bluebasic
8176 bytes free.
OK
SERIAL 9600,N,8,1,H
OK
WRITE #0,"A"
Not in direct

aanon4, 2015-02-07:
As far as I could tell reading the UART code from TI, they only support hardware handshaking.

mikewen, 2015-02-08:
I changed the UART code from TI , and compiled .hex files with software handshaking support:
Here is link of my Fork with .hex files for download:
https://github.com/mikewen/BlueBasic

aanon4, 2015-02-13:
Cool. Can you send me a patch?

aanon4, 2015-02-22:
Or a pull request perhaps?

closed 2015-03-25 by aanon4