aanon4/BlueBasic #13: Example on how to read from BLE slave?

mikewen, 2015-02-08:
Is it possible to add a simple example on how to read data from a BlueTooth BLE slave?

I was trying to use BlueBasic to read data from SensorTag, but only able to discover the device, not able to figure out how to read data from SensorTag.

Thanks a lot!

aanon4, 2015-02-13:
BlueBasic isn't a central device. If TI ever drop a 4.1 stack I'll switch it to be a Central+Peripheral device. While I'd like to support Central, you'd loose console access once the device switched.

aanon4, 2015-03-25:
Following up on this, it appears that TI will not be supporting a 4.1 stack on this generation of their chips, but only on the new ARM based ones.

closed 2015-03-25 by aanon4