aanon4/BlueBasic #14: Unable to use P2_1 through P2_4 for output

jtagen, 2015-02-18:
When I try to use P2_1 and later for output, get the message "Bad pin". Transcript of session below.

Thanks for creating this, I'm very excited!

BlueBasic CC2541/20150213105740/OAD v0.6
http://blog.xojs.org/bluebasic
7856 bytes free.
OK
PINMODE P2(0) OUTPUT
OK
P2(0) = 1
OK
PINMODE P2(1) OUTPUT
Bad pin
P2(1) = 1
Bad pin

jtagen, 2015-02-18:
Ah, I see this now, assuming it's specific to your hardware (debug and ?clock?).

I'm one of the unfortunates with no IAR license. Any chance you can provide a generic hex with all pins enabled?

aanon4, 2015-02-22:
True - I have specific hardware and excluded these in the builds. Might make sense to put them back for the basic 2540/1 builds and just exclude them for the Petra build (my hardware).

aanon4, 2015-02-22:
Just updates the builds. The non-Petra builds should now allow access to all P2 pins. Let me know how it looks.

jtagen, 2015-02-23:
Awesome, thank you!

closed 2015-02-23 by jtagen

jtagen, 2015-02-24:
Thanks again, this worked perfectly.