aanon4/BlueBasic #18: Using TI serial bootloader for BlueBasic

0xFACE, 2015-07-13:
Hello,
Now, with latest firmware it is possible to upgrade firmware for cheap HM-10/HM-11 module via UART, using serial bootloader protocol (http://processors.wiki.ti.com/index.php/SerialBootLoader)

Can I upload BlueBasic firmware this way into HM-10/11 module?

aanon4, 2015-07-25:
I've never investigated using the SerialBootLoader so I have no idea if it would work or if the HM-10/11 modules even use it. Maybe it could be made to work, and I'd be happy to accept a pull-request which does this, but I'm afraid I won't be doing it.

milkpirate, 2020-11-21:
@0xFACE My TI SmartRF Programmer does not seem to recognize my CC2541. So I am really interested in the serial method. Did you figure something out?

0xFACE, 2020-11-22:
Unfortunately not. 
Some of HM-xx modules from China have a serial firmware boot loader enabled. And it is possible to upgrade its firmware via serial interface. Not sure it is possible for the BlueBasic firmware. Ask this question here: https://github.com/kscheff/BlueBasic, this repository is under active development.

ozarchie, 2020-11-29:
bbconsole may help you.
It has been a while since I worked on it.
https://github.com/ozarchie/BlueBasic