aanon4/BlueBasic #19: How to connect do fresh flashed device ?

0xFACE, 2015-10-30:
Hello, I successfully flashed BLE HM-10 module with BlueBasic-CC2541.hex file.
But now I don't how to connect. 
I tried to scan for the BLE devices, but HM-10 module is not advertising any more.
Is it ok ? Can I use RX/TX pins to connect console?

Unfortunately I don't have IOS/ OSX device to use the  BlueBasicConsole from your repo.

crushedice2000, 2015-11-06:
I have the same problem with a <s>counterfeit</s> non-original HM-10.  Also there is a closed issue with the same problem: https://github.com/aanon4/BlueBasic/issues/10

0xFACE, 2015-11-11:
Tanks for the comment. I flashed another HM-11 module (smaller HM-10 version). 
It's advertising with "BASIC#7F". Now I can connect it with my phone and browse services.
So, it seems to be hardware issue with non-original HM-10 module (at first look one XTAL is missing comparing pictures). 

But still can't run any BASIC program.  What console to use for non IOS/ OSX OS'es ?

crushedice2000, 2015-11-11:
@0xFACE: In theory you could use one of these apps on Android (untested because I only have a non-original one):

MSMBLE
B-BLE
BLE Scanner
Bluetooth LE Scanner

crushedice2000, 2015-11-11:
@aanon4 (the main culprit ;-): Do you switch to the missing XTAL once the program begins? I can't figure out why the non-original HM-10 does extrange things... It starts advertising on power-up or reset and then stops/crashes... Could you test one of these?

0xFACE, 2015-11-23:
@crushedice2000
Yes, I can connect with these apps. But to write basic code I have to know which service(s) to use and how.

crushedice2000, 2015-11-23:
@0xFACE: There is a list of services. You can try them. However MSMBLE highlights in green/yellow the correct service.

0xFACE, 2015-12-20:
I didn't find any working software for Android, so I wrote small script to upload basic file from Linux:
https://github.com/0xFACE/BlueBasic-loader/blob/master/load

crushedice2000, 2015-12-20:
@0xFACE: Please check my PR.
[image]