aanon4/BlueBasic #20: loading BlueBasic using this method possible?

lorneb, 2015-11-28:
Hi excellent project, have been trying to load BlueBasic to a module for a couple of hours today.
this is the method I'm using https://github.com/RedBearLab/CCLoader

uses an arduino to act as a programmer.  
The module I am trying to update is a ble-cc41-a,  which has a cc2541 on it.

I was trying using the bin file for the cc2541 chip from you repo.  
it appears to flash ok, but cannot see any devices in the console.

Is the .bin from this repo comptable do you think?
Is there are way to check that code is running, debug output from module on serial ?

thanks in advance

crushedice2000, 2015-12-20:
If you can detect the device immediately after reset and then it stops advertising, it can be a problem with a missing xtal.

0xFACE, 2015-12-21:
Seems to be problem with missing 32kHz external crystal.
Interesting topic here: https://e2e.ti.com/support/wireless_connectivity/f/538/t/398466

crushedice2000, 2015-12-21:
@aanon4: You **should** build the binaries with this preprocessor directive:

CC2450: `XOSC32K_INSTALLED=FALSE`
CC2541: `OSC32K_CRYSTAL_INSTALLED=FALSE`

lorneb, 2015-12-22:
@aanon4 Could you please produce to more build outputs with the above suggestions?
TIA

aanon4, 2016-02-08:
Unfortunately I no longer have access to either the hardware or the tools to make any new builds. Sorry.

closed 2016-02-08 by aanon4

crushedice2000, 2016-02-08:
@aanon4: I figured it out. However, could you leave open this issue and simply edit README.md to explain that? Maybe I can fix this in a non-so-far future.

Thank you very much! Your project is amazing!