aanon4/BlueBasic #26: Question - Can a basic program be saved and survive a power off

bossesand, 2016-04-28:
I cannot find any information if it is possible to create a device with a permanent program that will run after power up.

How is that to be done?

JBtje, 2016-04-28:
https://github.com/JBtje/BlueBasic/wiki/_AUTORUN

I had some trouble with this command, seems not always to work as good as hoped for. In my implementation I let the software sens a RUN command each time it connects to the sensor.

Best regards,
Jeffrey

0xFACE, 2016-04-28:
I allways end my script with sentence 
```
AUTORUN ON
RUN
REBOOT

```

It works for me.

aanon4, 2016-05-16:
Programs are always written to FLASH so will always be there after power cycling. As noted by others, the AUTORUN ON command should make sure it runs after reboot. There's a short delay before the autorun to allow you to break in on a console in case there's a problem.

closed 2016-05-16 by aanon4