aanon4/BlueBasic #28: How to create characteristics with uint8 value respond

0xFACE, 2016-08-13:
BlueBasic is using 32-bit variables.  When I create characteristic for reading, the  response value is 32bit long. Is it possible to respond with uint8 values ?

According to the specification: **alert level** characteristic is  uint8:
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.alert_level.xml&u=org.bluetooth.characteristic.alert_level.xml

32bit response gives **invalid value** :

[image]

kscheff, 2016-12-06:
As far as I can see, it is always fixed to 4 bytes, big endian.
When using a array variable you can have different number of bytes, e.g. use DIM A(1) for a 1 byte characteristic.

0xFACE, 2016-12-07:
I tried to use arrays:
`GATT READ DIM A(1) ONREAD GOSUB 300`

But can't run the code, it exits with Error
Syntax seems to be incorrect.

kscheff, 2016-12-08:
@0xFACE try to using the below code.

You probably forgot to keep a idle loop running in order to keep the DIM A(1) scope. If the basic program terminates, A falls back to Int32 type and you get 4 bytes by reading.

-Kai

```
NEW
10 DIM A(1)

// setup advertsing
20 ADVERT GENERAL 
30 ADVERT "105a84de-40bd-428b-bf06-698e5e422cd9"
40 ADVERT END 
50 SCAN NAME "oneByteDemo"
60 SCAN END

// setup GATT profile

100 GATT SERVICE "105a84de-40bd-428b-bf06-698e5e422cd9" ONCONNECT GOSUB 400
110 GATT CHARACTERISTIC "00d8a2c6-65f3-40c2-a342-b4e5126f45f0" "sec"
120 GATT READ WRITE A ONREAD GOSUB 300
130 GATT END 

150 TIMER 0, 1000 REPEAT GOSUB 1000

// Idle loop keeping DIM A(1) alive
200 DELAY 10000
210 GOTO 200

// GATT read A
300 A(0) = T % 60
310 RETURN

// GATT Service on connect
400 TIMER 1, 2000 GOSUB 500
410 RETURN 

// welcome
500 PRINT "Hello World!"
510 RETURN 

// time tick
1000 T = T + 1
1010 RETURN

// this last return seems to be required
65529 RETURN

AUTORUN ON
END
REBOOT

```
[oneByte.bbasic.zip](https://github.com/aanon4/BlueBasic/files/640525/oneByte.bbasic.zip)

0xFACE, 2016-12-09:
Wow, great. 
Idle loop did the trick.
Thanks a lot ;-)