aanon4/BlueBasic #30: how to load basic programs on the chip?

pabloxid, 2016-09-13:
Is this "console" app. mean to load basic programs on the chip? Is there a Windows version?

0xFACE, 2016-09-14:
Yes, with console app you can load, list, edit programs on the chip. Unfortunately there is no Windows version.

crushedice2000, 2016-09-14:
@pabloxid: You can try [BlueBasic-loader](https://github.com/0xFACE/BlueBasic-loader). However it's for Linux, not Windows. Maybe you're interested in installing a Linux distribution like Ubuntu... I use it as my main operative system and I don't miss any functionality form M$ Windows 😈

ozarchie, 2016-09-14:
I think you can also use Botspine to do it. Although it complains that the
devices are not botspines, you can still load basic programs.

pabloxid, 2016-09-14:
Thanks for your answers.
I have a couple more questions:
1) how the console actually communicates with the device? Via Bluetooth?
2) Can I upload the firmware to the device with a basic program already included, using CC254x debugger?
3) there is some documentation on the console-device protocol?
Thank you again.

crushedice2000, 2016-09-14:
@pabloxid:

1. Yes, vía Bluetooth 4.0
2. I don't know, however somebody here could answer that.
3. Take a lookt to the script linked in my last message. You can see how it works. However, You cold así the autor for a detailed description.

ozarchie, 2016-10-18:
Unfortunately, OSX-Sierra completely breaks the console. Don't upgrade if you don't need to.

kscheff, 2016-11-08:
For OSX Sierra see #31

TheSmartGerman, 2018-08-07:
I'm currently searing for a app to download Basic Programms (Android / Windows would be fine).

I found this: https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal

you can configure the UUID Service, Sending and Receiving UUID.

The Console App out of the repository does the same:

```
140              if list[UUIDS.commsServiceUUID] != nil { 
141                self.inputCharacteristic = list[UUIDS.commsServiceUUID]!.characteristics[UUIDS.inputCharacteristicUUID] 
142                self.outputCharacteristic = list[UUIDS.commsServiceUUID]!.characteristics[UUIDS.outputCharacteristicUUID] 
```

TheSmartGerman, 2018-08-08:
You can use the Android Bluetooth Serial Terminal, it allows to use custom UUIDs: 
[](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=de)

It's working with the hex from this fork!

ozarchie, 2018-08-09:
I did this a while ago, but gave up on the CC2540 in favour of ESP8266
https://github.com/ozarchie/BlueBasic-loader

TheSmartGerman, 2018-08-09:
Well, I found a android solution. It's fine for me. 

I just mentioned it here. But beware using the branch from kscheff, does have a problem with the cc2541 build: https://github.com/kscheff/BlueBasic/issues/8

The ESP8266 / ES32 a known but right know no option....