aanon4/BlueBasic #31: OSX 10.12 Sierra fails

kscheff, 2016-11-08:
BlueBasic under OSX Sierra fails:
- after connecting File/Load to device doesn't show file selector
- pasting code to the console fails to complete

xcode 8.1 doesn't support Swift 1.2 anymore. In an attempt to fix the above issue, I imported the project to Xcode 8.1 and converted it to Swift 3. Then I fixed the upload issue by reducing the upload data chunks from 64 down to 20 bytes.

See fork here:
https://github.com/kscheff/BlueBasic/commit/6de2ee0b86112d801c0c7ff3a3b286cada15b31b

Netoperz, 2016-12-08:
Your fork also cannot connect, so i assume that this project is useless at this momment as there is no way to upload the actual program to the chip.

So for this moment the bluebasic cannot be used on windows, on osx, and i do not have place to install third OS just for one program... LOL.

Sometimes it is better to first read issues and than wiki :)

You will loose less time.

I've build the device, bougt CC debuger flashed firmware and installed the app from appstore, and EPIC fail.

But thanx for your effort and good work.

The problems with OS X arfe year by year the same.

Those bustards are changing things, and year by year there are poroblems with USB to Serial bridges, USB controllers. Compillers, and now as i've heard from fellow programmers changing swift from 2.3 to 3, or something like that, so many apps needs actualization.

I loved OS X, now i Hate it. I bougt old lenovo Thinkpad x240 and with win 10 for coding everything works fine.

I'm 37 years old, and this is my firs windows instalation cince my birth. Linux is fine but please try to find IAR compiler for that platform....

Waiting for the fix.

kscheff, 2016-12-08:
Sorry to hear all the issues... Sometimes its hard to get the chain working, especially when its so complex with all "beta" type of software. Trust me I walked through that as well. Finding an IAR Complier is not the issue - its the price of the compiler which makes it useless for such projects, unless you have a well funded one.

I too had connection issue on the BlueBasic OSX console, even after it was working. Tim supplied a nice script resetting the bluetooth cache in the /utils directory of the project. Executing this helped once. Another source of trouble is a messed up BLE stack, here it helps to switch Bluetooth off/on in the main OSX settings. I am using an early 2011 MacBook Pro  with an CSR8510 A10 Bluetooth dongle sticking in one USB port. I am using macOS Sierra 10.12.2 Beta at this moment.

Another option to get the device going is to use the console on iOS, but then you would need to compile the project and send it over to an iPhone.

Netoperz, 2016-12-08:
strange, now your and the orginal version works. just updated the Xcode , weird ...

Netoperz, 2016-12-08:
Ok, so ... the app from appstore on my laptop does not work.
It works partially, it can discover the bluebasic device, and it can connect, but the code is not uploaded or executed, i'm getting ok after hello world test, but no hello world answer printed.

The app from your fork works with hello world, and prints back the hello world.

The beakon example is not working. so could you test the iBeakon example and tell if it works ?

I'm using Beacon scanner on android phone , exactly samsung A5 wit android 6.0 to find the beakon, and it sees other beakons around but not mine.

kscheff, 2016-12-09:
@netoperz Great to hear progress. So I am not alone here. I too use this Firmware for a project, running an Solar charging data logger for camping mobiles. As far as I remember the iBeacon example worked.

I have not looked into how to lock the access, there is a way for PIN access. But I am not shure if it works and second if it is only for induvidual privacy... 

One other way would be to use the ONCONNECT GOSUB like in Hello World, but terminate the connection immediately by GATT CLOSE command. Then there is only access in the first 5 seconds before the BASIC program starts. So if you can make shure no unsurhorized  power cycle happens, then this would be a sufficient lock. However this locks you out for updates in case you have no physical access.

You might consider a special function with sequence and a hardware connection to generate a software driven reset, resulting in the 5 second access window.

This all is not a very secure meachanism, but might prevent someone driving by and messing around with your hardware. If you need more robust system, then this can be archived with special Firmware. I have developed such on the cc2541 chip with custom firmware. This is a more time consuming task, especially if you are not familiar with all the tools and chip internals.

Netoperz, 2016-12-09:
Good to hear that.

CC2540 is in use because i have been developing a serial to BT 2.0 bridge using HC06 bt 2.0 module. and i occasionally ordered HC-08 that is bt smart from the same seller (a few pieces) so they were in my desk. Now i need the iBeakons for estimating the distance to the object, and also to send HIGH signal to BT smart device aka iBeakon to for example  drive analog beeper when needed, when the object with iBeakon is farther than 10 meters the software on phone should trigger the alarm on beacon. So first choice was the hardware i have.

In future i think CC2541 will be used, right now i just test the ability to use BT 4.0 and what can be done and what does not.

I would love to look on the code for 2541 to understand how to get what i need . So if i will decide that i need my own build than i'll let you know and many you will be willing to share your work, we can share ours in exchange also, and a bottle of wine or good beer if you plan to visit Poland , especially Cracow (south of poland) 

Cheers mate, and thank you for your effort. that is priceless. :)