aanon4/BlueBasic #33: Writing Array fails

kscheff, 2016-11-11:
Hello,

I understand that this repro is kind of "dead" since Tim has left the project. However this is the only place to "hope" for exchange of experience... maybe someone is reading here.

Writing an array to a file works, however after the write command the execution stalls.
Here is my program list

```
10 OPEN 0, READ "A"
15 FOR I = 1 TO 10
20  READ #0, B
30  PRINT I,",",B
40 NEXT I
60 RETURN 
100 OPEN 0, TRUNCATE "A"
110 DIM U(10)
120 WRITE #0, U
140 CLOSE 0
150 RETURN 

goto 100
Bad expression
>> 140 CLOSE 0
```

I can read the file successfully by walking to its elements via:

```
run
1,0
2,0
3,0
4,0
5,0
6,0
7,0
8,0
9,0
10,0
OK
```

The issue I am fighting is that I wanted to implement an IoT data logger for solar cells mounted on the camping car. Therefore I need a way to store and access some historic values. Doing by writing single vars wastes a lot of memory, e.g. each var occupies 8 bytes (examine the reported memory). So I was looking to write an array instead. It looks like that each array element consumes 1 byte.

Is there anybody out there that have successfully used these functions?

Thanks
Kai

kscheff, 2016-11-15:
I found a workaround for the issue by not closing the file after WRITE and instead place a RETURN after it. My write routine runs then in a timer interrupt.

kaiateic, 2017-01-03:
Hi Kai,

I think, in some cases, the interpreter goes on to the next command before it is finished with the current command. File IO might be one of them. Try putting DELAY 1000 before your CLOSE command and see if that fixes it. 

BTW, we are considering porting BlueBasic to a more modern platform (cc2650). TI seems to be pushing that micro. Do you want to get involved?

Cheers,
Kai

kscheff, 2017-01-03:
@kaiateic this might be a test worth...
be aware there are other issues with the DELAY command when in a GOSUB during a GATT access.

I have looked into the cc2650 for some projects, so far ready modules with this device are rare and expensive (thus I can imagine TI is more interested in pushing this device to get out of the low margin cc2541 modules). However the cc2650 has some nice features like lower power and more processing.

I have the dev kit in my drawer.... so there is interest on my side.

Cheers
Kai

kaiateic, 2017-01-03:
Great! 

Let's take this conversation off line. Email me at kai@eic.nu

ozarchie, 2017-01-07:
Hi Kai,
Have a look at alibaba-radioland for the cc256x devices. They are now in
the  $2-4 range. I would be interested in helping port/test Bluebasic. I am
currently using the existing code in hydroponic automation. Cheers,
ozarchie

kaiateic, 2017-01-08:
Hi Ozarchie
Email me at the address above and lets take this offline.
Kai

kscheff, 2017-01-08:
@ozarchie 
I couldn't find a module with the CC256x device. Currently I am getting CC2541 modules JPY-8 for < $2.30 in quantities of 10. If you a link for a CC256x device please send it to me.

TX Kai

ozarchie, 2017-01-09:
Hi Kai,
Try http://en.szrfstar.com. I  have used them for 254x but not 256x
Cheers, John

kaiateic, 2017-01-09:
Looks like there are a lot more FCC certified modules available using Nordic chips than TI. (If it doesn't have shielding, it's not fully certified.) Might be a better idea to port to Nordic instead?
Kai

kscheff, 2017-08-30:
Bugfix see here:
https://github.com/kscheff/BlueBasic/commit/447765a015cdad66632670f79012363768bb4b21